meshtastic/Meshtastic-Android #1448: [Bug]: Import and export config gone

FredB1959, 2024-12-04:
### What happened?

In radio configuration the import and export config is no longer availible.

### App Version

2.5.12

### Phone

Pixel 7, Android 15

### Device

Heltec V3

### Firmware

2.5.15.f2ee0df

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-12-05:
this is fixed in 2.5.13 (https://github.com/meshtastic/Meshtastic-Android/commit/76454cc802f6fa9705c02042c8e9cd79190ec8f7)

closed 2024-12-05 by andrekir

sevenbitbyte, 2024-12-15:
When will this be release to Google Play and f-droid? With how frequently certain radios reset and the now active use of admin management its a real show stopper when you can't quickly restore from backups.