meshtastic/Meshtastic-Android #1451: [Feature Request]: add a a tag or special character to allow custom quick replies to a message

Nestpebble, 2024-12-07:
### Contact Details

NomDeTom

### Tell us your idea.

I'd like the app to use an unprintable character to turn the following word into a hyperlink that can be tapped to auto-fill a reply.

The primary purpose would be replying to a bot, auto responder or BBS with contextual replies, e.g. (n)ext, (b)ack, (m)ore, (d)elete, etc.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rndmcnlly, 2025-01-16:
# Slight generalization

I like the idea of this quick reply feature, and I have an idea for how to make it more accessible and generic.

Rather than looking for a magic (perhaps unprintable) character, let's look for spans of text that follow the specific convention of "(n)ext". If the user enables the quick replies UI option in the app, the app can scan for and make tappable spans that appear to be quick-reply options in the same way that it detects and upgrades URLs.

Here's a slightly over-generalized way of detecting these with a regex:

```java
String pattern = "\\w*\\([\\w]{1,3}\\)\\w+";

// These will match
"(b)ack"
"(4)email"
"re(t)urn"

// These won't match
"re(test)turn"  // too many chars in parentheses
"re()turn"      // empty parentheses
"re(t)"         // no chars after parentheses
```

In this design, no special keyboard is needed for human users to offer one another quick replies.

# Major generalization

Interpret the message as Markdown text (applying formatting like bold, italics, lists, whatever), and give special behavior to links with no target. Treat a no-target link as something that should be copied into the message composition field. For example `[web](http://example.com/)` would display as a link that tries to open in a browser but `[back]()` renders a button (or link like [back]()) that appends "back" to the current message.

closed 2025-05-24 by github-actions[bot]

NomDeTom, 2025-05-25:
Woohoo! So back!