meshtastic/Meshtastic-Android #1453: [Feature Request]: Add option to hide/disable emoji reactions

sycophantic, 2024-12-09:
### Tell us your idea.

I'd like the app to behave before this emoji reactions feature was added. Please add an option to disable seeing them.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2024-12-09 by garthvh

sycophantic, 2024-12-09:
Guess it's time for a fork.

sycophantic, 2024-12-09:
It's unneeded mesh traffic.

RCGV1, 2024-12-09:
A single emoji is much less data than a text message. Also this feature has already existed on the iOS app now android users can also send emoji reactions.

b8b8, 2024-12-09:
It's also a feature in every other messaging app.

sycophantic, 2024-12-09:
Metastatic isn't "every other messaging app". We don't need emoji reactions for emergency communications.

deadman96385, 2024-12-09:
It would be nice if it was a toggle, so it can be removed to not clutter up space in the UI. Default the switch to be on to not break current behavior and expectations. I don't agree with the whole just because other apps have it you need it as well philosophy. But I can see why some people may like the feature.

sycophantic, 2024-12-09:
> It would be nice if it was a toggle, so it can be removed to not clutter up space in the UI. Default the switch to be on to not break current behavior and expectations. I don't agree with the whole [...]

That's all we need is the toggle.

sycophantic, 2024-12-10:
Can someone explain how this is ok? This is why I want a toggle.
[image]

garthvh, 2024-12-10:
You are using a global public channel, you can't control what he content of messages sent on a globally public channel. This is a message like any other, and is not new functionality.

sycophantic, 2024-12-10:
@garthvh why is so hard to just add a toggle in the application?