meshtastic/Meshtastic-Android #1462: [Bug]: App reports "Error Disconnect" incorrectly

ctmalan, 2024-12-13:
### Contact Details

ctmalan@gmail.com

### What happened?

When the acknoledge message from the T-Echo connected via BLE does not arrive quicly enough:  "Error Disconnected" is reported.  The update is successful when the device reboots.

This does not happen with all the commands.

### App Version

2.5.13

### Phone

Samsung Galaxy A04e

### Device

UF2 Bootloader 0.6.1-2-g1224915 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90f) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: LilyGo T-Echo
Board-ID: nRF52840-TEcho-v1
SoftDevice: S140 version 6.1.1
Date: Feb 10 2024

### Firmware

Various:  
2.5.11
2.5.14
2.5.15
2.5.16

Did not notice this with 2.3.9

### Relevant log output

```shell
DEBUG | 13:51:46 269 [Power] Battery: usbPower=1, isCharging=1, batMv=4883, batPct=100
DEBUG | 13:51:47 270 [GPS] Took 7s to get lock
DEBUG | 13:51:47 270 [GPS] Predict 4s to get next lock
DEBUG | 13:51:48 270 [GPS] 115s until next search
DEBUG | 13:51:48 270 [GPS] gps_update_interval >= 14s needed to justify hardsleep
INFO  | 13:51:48 270 [GPS] GPS power state move from ACTIVE to HARDSLEEP
DEBUG | 13:51:48 270 [GPS] Publish pos@675c3bf4:2, hasVal=1, Sats=7, GPSlock=1
DEBUG | 13:51:48 270 [GPS] New GPS pos@675c3bf4:3 lat=-34.106460 lon=18.828431 alt=-1 pdop=2.67 track=0.00 speed=0.01 sats=7
DEBUG | 13:51:48 270 [GPS] onGPSChanged() pos@675c3bf4 time=1734097908 lat=-341064600 lon=188284311 alt=-1
INFO  | 13:51:48 270 [GPS] updatePosition LOCAL pos@675c3bf4 time=1734097908 lat=-341064600 lon=188284311 alt=-1
DEBUG | 13:51:48 270 [GPS] Set local position: lat=-341064600 lon=188284311 time=1734097908 timestamp=1734097908
DEBUG | 13:51:48 270 [GPS] Node status update: 2 online, 2 total
DEBUG | 13:52:06 289 [Power] Battery: usbPower=1, isCharging=1, batMv=4895, batPct=100
DEBUG | 13:52:07 290 [Screen] determineMode():
DEBUG | 13:52:07 290 [Screen] refresh=FAST, reason=BACKGROUND_USES_FAST, fastRefreshCount=2, frameFlags=0x1
DEBUG | 13:52:07 290 [Screen] Update E-Paper
DEBUG | 13:52:08 291 [Screen] done
INFO  | 13:52:08 291 [DeviceTelemetry] Send: air_util_tx=0.079583, channel_utilization=0.000000, battery_level=101, voltage=4.895000, uptime=291
DEBUG | 13:52:08 291 [DeviceTelemetry] Partially randomized packet id 1874712683
DEBUG | 13:52:08 291 [DeviceTelemetry] updateTelemetry LOCAL
DEBUG | 13:52:08 291 [DeviceTelemetry] Node status update: 2 online, 2 total
INFO  | 13:52:08 291 [DeviceTelemetry] Send packet to phone
INFO  | 13:52:08 291 Tell client we have new packets 24
INFO  | 13:52:08 291 BLE notify fromNum
DEBUG | 13:52:08 291 FromRadio=STATE_SEND_PACKETS
DEBUG | 13:52:08 291 phone downloaded packet (id=0x6fbdd86b fr=0xc9a421a9 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1734097928 priority=10)
DEBUG | 13:52:26 309 [Power] Battery: usbPower=1, isCharging=1, batMv=4902, batPct=100
INFO  | 13:52:28 311 toRadioWriteCb data 0x20020c3a, len 28
DEBUG | 13:52:28 311 New ToRadio packet
DEBUG | 13:52:28 311 PACKET FROM PHONE (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP priority=70)
DEBUG | 13:52:28 311 Enqueued local (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP rxtime=1734097948 priority=70)
[... 206 lines cut ...]
DEBUG | 13:52:55 338 [Router] FromRadio=STATE_SEND_PACKETS
DEBUG | 13:52:55 338 [Router] Rx someone rebroadcasting for us (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priorit
DEBUG | 13:52:55 338 [Router] Didn't find pending packet
DEBUG | 13:52:55 338 [Router] Add packet record (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
DEBUG | 13:52:55 338 [Router] handleReceived(REMOTE) (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
DEBUG | 13:52:55 338 [Router] Module 'neighborinfo' wantsPacket=1
INFO  | 13:52:55 338 [Router] Received neighborinfo from=0xc9a421a9, id=0x26ce786f, portnum=5, payloadlen=2
DEBUG | 13:52:55 338 [Router] Module 'neighborinfo' considered
DEBUG | 13:52:55 338 [Router] Module 'routing' wantsPacket=1
INFO  | 13:52:55 338 [Router] Received routing from=0xc9a421a9, id=0x26ce786f, portnum=5, payloadlen=2
DEBUG | 13:52:55 338 [Router] Routing sniffing (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
DEBUG | 13:52:55 338 [Router] Received a ACK for 0x813c560c, stopping retransmissions
DEBUG | 13:52:55 338 [Router] Delivering rx packet (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
DEBUG | 13:52:55 338 [Router] Update DB node 0xc9a421a9, rx_time=1734097975
DEBUG | 13:52:55 338 [Router] Forwarding to phone (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
DEBUG | 13:52:55 338 [Router] Module 'routing' considered
INFO  | 13:52:55 338 Tell client we have new packets 32
INFO  | 13:52:55 338 BLE notify fromNum
DEBUG | 13:52:55 338 FromRadio=STATE_SEND_PACKETS
DEBUG | 13:52:55 338 phone downloaded packet (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ctmalan, 2024-12-13:
DEBUG | 13:51:46 269 [Power] Battery: usbPower=1, isCharging=1, batMv=4883, batPct=100
DEBUG | 13:51:47 270 [GPS] Took 7s to get lock
DEBUG | 13:51:47 270 [GPS] Predict 4s to get next lock
DEBUG | 13:51:48 270 [GPS] 115s until next search
DEBUG | 13:51:48 270 [GPS] gps_update_interval >= 14s needed to justify hardsleep
INFO  | 13:51:48 270 [GPS] GPS power state move from ACTIVE to HARDSLEEP
DEBUG | 13:51:48 270 [GPS] Publish pos@675c3bf4:2, hasVal=1, Sats=7, GPSlock=1
DEBUG | 13:51:48 270 [GPS] New GPS pos@675c3bf4:3 lat=-34.106460 lon=18.828431 alt=-1 pdop=2.67 track=0.00 speed=0.01 sats=7
DEBUG | 13:51:48 270 [GPS] onGPSChanged() pos@675c3bf4 time=1734097908 lat=-341064600 lon=188284311 alt=-1
INFO  | 13:51:48 270 [GPS] updatePosition LOCAL pos@675c3bf4 time=1734097908 lat=-341064600 lon=188284311 alt=-1
DEBUG | 13:51:48 270 [GPS] Set local position: lat=-341064600 lon=188284311 time=1734097908 timestamp=1734097908
DEBUG | 13:51:48 270 [GPS] Node status update: 2 online, 2 total
DEBUG | 13:52:06 289 [Power] Battery: usbPower=1, isCharging=1, batMv=4895, batPct=100
DEBUG | 13:52:07 290 [Screen] determineMode():
DEBUG | 13:52:07 290 [Screen] refresh=FAST, reason=BACKGROUND_USES_FAST, fastRefreshCount=2, frameFlags=0x1
DEBUG | 13:52:07 290 [Screen] Update E-Paper
DEBUG | 13:52:08 291 [Screen] done
INFO  | 13:52:08 291 [DeviceTelemetry] Send: air_util_tx=0.079583, channel_utilization=0.000000, battery_level=101, voltage=4.895000, uptime=291
DEBUG | 13:52:08 291 [DeviceTelemetry] Partially randomized packet id 1874712683
DEBUG | 13:52:08 291 [DeviceTelemetry] updateTelemetry LOCAL
DEBUG | 13:52:08 291 [DeviceTelemetry] Node status update: 2 online, 2 total
INFO  | 13:52:08 291 [DeviceTelemetry] Send packet to phone
INFO  | 13:52:08 291 Tell client we have new packets 24
INFO  | 13:52:08 291 BLE notify fromNum
DEBUG | 13:52:08 291 FromRadio=STATE_SEND_PACKETS
DEBUG | 13:52:08 291 phone downloaded packet (id=0x6fbdd86b fr=0xc9a421a9 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1734097928 priority=10)
DEBUG | 13:52:26 309 [Power] Battery: usbPower=1, isCharging=1, batMv=4902, batPct=100
INFO  | 13:52:28 311 toRadioWriteCb data 0x20020c3a, len 28
DEBUG | 13:52:28 311 New ToRadio packet
DEBUG | 13:52:28 311 PACKET FROM PHONE (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP priority=70)
DEBUG | 13:52:28 311 Enqueued local (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] Add packet record (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] handleReceived(REMOTE) (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] Module 'Admin' wantsPacket=1
INFO  | 13:52:28 311 [Router] Received Admin from=0x0, id=0x813c560a, portnum=6, payloadlen=2
INFO  | 13:52:28 311 [Router] Handle admin payload 7
INFO  | 13:52:28 311 [Router] Client got module config
INFO  | 13:52:28 311 [Router] Get module config: MQTT
DEBUG | 13:52:28 311 [Router] Set admin key to  3e ee b7 fc d9 9b 17 36

DEBUG | 13:52:28 311 [Router] Partially randomized packet id 1758558316
INFO  | 13:52:28 311 [Router] Asked module 'Admin' to send a response
DEBUG | 13:52:28 311 [Router] Module 'neighborinfo' wantsPacket=1
INFO  | 13:52:28 311 [Router] Received neighborinfo from=0x0, id=0x813c560a, portnum=6, payloadlen=2
DEBUG | 13:52:28 311 [Router] Module 'neighborinfo' considered
DEBUG | 13:52:28 311 [Router] Module 'routing' wantsPacket=1
INFO  | 13:52:28 311 [Router] Received routing from=0x0, id=0x813c560a, portnum=6, payloadlen=2
DEBUG | 13:52:28 311 [Router] Routing sniffing (id=0x813c560a fr=0x00000000 to=0xc9a421a9, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 WANTRESP rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] Another module replied to this message, no need for 2nd ack
DEBUG | 13:52:28 311 [Router] Module 'routing' considered
DEBUG | 13:52:28 311 [Router] Send response (id=0x68d1786c fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=6 requestId=813c560a rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] Update DB node 0xc9a421a9, rx_time=1734097948
DEBUG | 13:52:28 311 [Router] Enqueued local (id=0x68d1786c fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=6 requestId=813c560a rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] Rx someone rebroadcasting for us (id=0x68d1786c fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=6 requestId=813c560a rxtime=1734097948 priorit
DEBUG | 13:52:28 311 [Router] Didn't find pending packet
DEBUG | 13:52:28 311 [Router] Add packet record (id=0x68d1786c fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=6 requestId=813c560a rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] handleReceived(REMOTE) (id=0x68d1786c fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=6 requestId=813c560a rxtime=1734097948 priority=70)
DEBUG | 13:52:28 311 [Router] Module 'Admin' wantsPacket=1
INFO  | 13:52:28 311 [Router] Received Admin from=0xc9a421a9, id=0x68d1786c, portnum=6, payloadlen=87
DEBUG | 13:52:28 311 [Router] Allow admin response message
INFO  | 13:52:28 311 [Router] Handle admin payload 8
INFO  | 13:52:28 311 [Router] Client received a get_module_config response
[... cut ...]
DEBUG | 13:52:55 338 [Router] Forwarding to phone (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)
DEBUG | 13:52:55 338 [Router] Module 'routing' considered
INFO  | 13:52:55 338 Tell client we have new packets 32
INFO  | 13:52:55 338 BLE notify fromNum
DEBUG | 13:52:55 338 FromRadio=STATE_SEND_PACKETS
DEBUG | 13:52:55 338 phone downloaded packet (id=0x26ce786f fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c560c rxtime=1734097975 priority=120)

ctmalan, 2024-12-13:
You will be able to align this with the posted bug report:

DEBUG | 13:27:53 521 [Screen] determineMode():
DEBUG | 13:27:53 521 [Screen] refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, fr                                                 ameFlags=0x1
DEBUG | 13:28:01 530 [Power] Battery: usbPower=1, isCharging=1, batMv=4914, batPct=100
INFO  | 13:28:02 530 [DeviceTelemetry] Send: air_util_tx=0.132306, channel_utilization=0.000000, battery_level=101, voltage=4.914000, uptime=530
DEBUG | 13:28:02 530 [DeviceTelemetry] Partially randomized packet id 3720432633
DEBUG | 13:28:02 530 [DeviceTelemetry] updateTelemetry LOCAL
DEBUG | 13:28:02 530 [DeviceTelemetry] Node status update: 2 online, 2 total
INFO  | 13:28:02 530 [DeviceTelemetry] Send packet to phone
INFO  | 13:28:02 530 Tell client we have new packets 47
INFO  | 13:28:02 530 BLE notify fromNum
DEBUG | 13:28:02 530 FromRadio=STATE_SEND_PACKETS
DEBUG | 13:28:02 530 phone downloaded packet (id=0xddc14bf9 fr=0xc9a421a9 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1734096482 priority=10)
DEBUG | 13:28:02 531 [RadioIf] Lora RX (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:02 531 [RadioIf] Packet RX: 665ms
DEBUG | 13:28:02 531 [Router] Add packet record (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:02 531 [Router] Use channel 0 (hash 0x8)
DEBUG | 13:28:02 531 [Router] Expand short PSK #1
DEBUG | 13:28:02 531 [Router] Use AES128 key!
DEBUG | 13:28:02 531 [Router] decoded message (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:02 531 [Router] handleReceived(REMOTE) (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:02 531 [Router] Module 'position' wantsPacket=1
INFO  | 13:28:02 531 [Router] Received position from=0xa24db08f, id=0x9b6c886d, portnum=3, payloadlen=36
DEBUG | 13:28:02 531 [Router] POSITION node=a24db08f l=36 lat=-341082112 lon=188252160 msl=15 hae=0 geo=0 pdop=211 hdop=0 vdop=0 siv=9 fxq=0 fxt=0 pts=0 time=1734096481
DEBUG | 13:28:02 531 [Router] Ignore time from mesh because we have a GPS, RTC, or Phone/NTP time source in the past day
INFO  | 13:28:02 531 [Router] updatePosition REMOTE node=0xa24db08f time=1734096481 lat=-341082112 lon=188252160
DEBUG | 13:28:02 531 [Router] Node status update: 2 online, 2 total
DEBUG | 13:28:02 531 [Router] Module 'position' considered
DEBUG | 13:28:02 531 [Router] Module 'neighborinfo' wantsPacket=1
INFO  | 13:28:02 531 [Router] Received neighborinfo from=0xa24db08f, id=0x9b6c886d, portnum=3, payloadlen=36
DEBUG | 13:28:02 531 [Router] Module 'neighborinfo' considered
DEBUG | 13:28:02 531 [Router] Module 'routing' wantsPacket=1
INFO  | 13:28:02 531 [Router] Received routing from=0xa24db08f, id=0x9b6c886d, portnum=3, payloadlen=36
DEBUG | 13:28:02 531 [Router] Routing sniffing (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3)
INFO  | 13:28:02 531 [Router] Rebroadcast received floodmsg
DEBUG | 13:28:02 531 [Router] Expand short PSK #1
DEBUG | 13:28:02 531 [Router] Use AES128 key!
DEBUG | 13:28:02 531 [Router] enqueue for send (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3 priority=6
DEBUG | 13:28:02 531 [Router] txGood=8,txRelay=1,rxGood=10,rxBad=0
DEBUG | 13:28:02 531 [Router] rx_snr found. hop_limit:2 rx_snr:6.500000
DEBUG | 13:28:02 531 [Router] rx_snr found in packet. Setting tx delay:2772
DEBUG | 13:28:02 531 [Router] Delivering rx packet (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:02 531 [Router] Update DB node 0xa24db08f, rx_time=1734096482
DEBUG | 13:28:03 531 [Router] Forwarding to phone (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:03 531 [Router] Module 'routing' considered
INFO  | 13:28:03 531 [Router] MQTT onSend - Not forwarding packet due to DontMqttMeBro flag
INFO  | 13:28:03 531 Tell client we have new packets 48
INFO  | 13:28:03 531 BLE notify fromNum
DEBUG | 13:28:03 531 FromRadio=STATE_SEND_PACKETS
DEBUG | 13:28:03 531 phone downloaded packet (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3)
DEBUG | 13:28:05 534 [RadioIf] Started Tx (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3 priority=64)
DEBUG | 13:28:05 534 [RadioIf] Packet TX: 665ms
DEBUG | 13:28:06 534 [RadioIf] Completed sending (id=0x9b6c886d fr=0xa24db08f to=0xffffffff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1734096482 rxSNR=6.5 rxRSSI=-39 hopStart=3 priority=
DEBUG | 13:28:21 550 [Power] Battery: usbPower=1, isCharging=1, batMv=4912, batPct=100
DEBUG | 13:28:23 552 [Screen] determineMode():
DEBUG | 13:28:23 552 [Screen] refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x1
INFO  | 13:28:28 556 [Position] Send pos@675c3644:6 to mesh (wantReplies=0)
DEBUG | 13:28:28 556 [Position] cancelSending id=0x11083e1, removed=0
[... cut ...]
DEBUG | 14:02:45 128 [Router] Forwarding to phone (id=0xc352ebe0 fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c5614 rxtime=1734098565 priority=120)
DEBUG | 14:02:45 128 [Router] Module 'routing' considered
INFO  | 14:02:45 128 Tell client we have new packets 26
INFO  | 14:02:45 128 BLE notify fromNum
DEBUG | 14:02:45 128 FromRadio=STATE_SEND_PACKETS
DEBUG | 14:02:45 128 phone downloaded packet (id=0xc352ebe0 fr=0xc9a421a9 to=0xc9a421a9, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=813c5614 rxtime=1734098565 priority=120)

andrekir, 2024-12-13:
this issue is being tracked at https://github.com/meshtastic/firmware/issues/5068

closed 2024-12-13 by andrekir

ctmalan, 2024-12-13:
> <https://github.com/meshtastic/Meshtastic-Android/issues/1462> as not

Please note that #5068 consists of two issues:

1  Flashing firmware throws an error almost every time new firmware is
drag-and-dropped into the folder - this is a firmware update bug
2  Config update throwing "Error Disconnect" in some cases - this is a
configuration change bug

These should have separate tickets.

Please confirm.  I might be misreading the bug reports.
--
c-: