meshtastic/Meshtastic-Android #1467: Traceroute loading bar

cracky22, 2024-12-15:
### Contact Details

martinblieninger2208@gmail.com

### Tell us your idea.

Loading bar until the 30 seconds have elapsed until the next trace route

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GUVWAF, 2024-12-15:
You already opened the same issue (#1422), which was closed as "not planned". 
The reason why it's probably not planned is because the 30-second limit is not there because it means the previous traceroute failed, but it's there to limit traffic. The loading bar shouldn't be an encouragement to try a next one as soon as possible.

cracky22, 2024-12-15:
Thanks for the answer and explanation of how you imagine it, yes, I created the issue twice because it was closed without comment!

closed 2024-12-15 by cracky22