meshtastic/Meshtastic-Android #1474: [Bug]: Message text area is not always cleaned when sending

VigibotDev, 2024-12-15:
### Contact Details

admin@serveurperso.com

### What happened?

Message text area is not deleted some time on sending (it must be cleaned all time)

This looks exactly like the bug you can get in web development, A timing problem on slower or faster machines if the delete event is done on the button press rather than a callback after submit.

[image]

### App Version

2.5.14 last google beta program APK

### Phone

Samsung galaxy s24 ultra

### Device

RAK WisBlock 4631

### Firmware

2.5.15 lastest Stable firmware

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

weirdal3333, 2024-12-15:
What keyboard are you using? I'm using FUTO and have experienced this bug in other applications as well as Meshtastic!
Not sure if it's an Android feature change, the app itself, or the keyboard!

VigibotDev, 2024-12-16:
I use "Microsoft SwiftKey AI Keyboard" as my default keyboard on all android device and I have never had this bug on any other application. Android must completely isolate the operation of the application by the API abstraction layer.
Knowing that the send button belongs to the application and that the text entry works well.
I do not use any special function with Meshtastic, just a press on the completely standard touch keys, and possibly the one that offers a choice of completion in case of typos.

VigibotDev, 2024-12-16:
I just discovered that if the text ends with a space, it is cleaned, if it does not end with a space, it is not cleaned.
The Standard Letters and Word Completion buttons do not change this behavior.

https://github.com/user-attachments/assets/b509857c-25a5-4404-a6ae-9d7c81711243

VigibotDev, 2024-12-16:
It's more subtle than that, the cursor must also be positioned at the end after the space for the message to be cleaned! It still looks like a cleaning timing problem... for some reason there must be impacts that I can't explain...

VigibotDev, 2024-12-21:
I confirm solved in 2.5.15 beta update. Thanks for your work

ServeurpersoCom, 2025-06-21:
I confirm this issue come back in Android v2.6.19 (Google beta program)
- All hardware
- Firmware last stable/beta
- Bluetooth

jamesarich, 2025-06-21:
i am unable to reproduce in latest builds. may need some extra looking though.

commit 5693874d (2024-12-17): fix: message text is not always cleaned when sending

fixes #1474
PR #2231 (2025-06-23): Refactor message input to use BasicTextField2
This commit refactors the message input field in the Message screen to use the new `BasicTextField2` from Jetpack Compose Foundation.

Provides cleaner handling for clearing state.

The key changes include:
- Replacing `MutableState<TextFieldValue>` with `TextFieldState`.
- Using `rememberTextFieldState` to manage the input field's state.
- Updating text manipulation from `messageInput.value = TextFieldValue(...)` to `messageInput.setTextAndPlaceCursorAtEnd(...)` and `messageInput.clearText()`.
- Simplifying keyboard event handling for sending mess [... cut]