meshtastic/Meshtastic-Android #1479: [Feature Request]: Add short Name under emojis if choose.

Dtrieb, 2024-12-15:
### Contact Details

dtrieb

### Tell us your idea.

If you change the short Name to a emoji in the android app you can't distinguish people (in the chat list) if they use the same emojis.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-16:
When you change your shortname to an emoji that is now your shortname.

garthvh, 2024-12-16:
What you really want is something like this on iOS
[image]

Dtrieb, 2024-12-16:
Yes because if I am on a full MQTT Channel there will be a huge chance to have 2 or more same emojis.

commit 33f86dcf (2024-12-17): feat: add user `longName` and `userId` to messages

closes #1479
closed 2024-12-17 by andrekir