meshtastic/Meshtastic-Android #1482: "Error Disconnected" for Android on T-Echo

ctmalan, 2024-12-17:
### Discussed in https://github.com/meshtastic/firmware/discussions/5537

<div type='discussions-op-text'>

<sup>Originally posted by **ctmalan** December  9, 2024</sup>
I received 2 x T-Echos with firmware 2.3.9"
`Board-ID: nRF52840-TEcho-v1
SoftDevice: S140 version 6.1.1
Date: Feb 10 2024
`

Everything seemed fine except for the traceroute.  I therefore did a factory reset download via the serial port and upgraded the firmware to 2.5.14.

The region upgrade reset seemed fine and the traceroute now worked but any "Radio Config" change now reports:
 `Error
Disconnected`

The change is implemented.

How do I fix this?  If I cannot is this enough info for a gug report?

c-:</div>

To recreate:  go to radio settings - MQTT - change topic and send.

After further investigation I found that I needed to go back to 2.5.11 to get the nodes to show up in meshmap.net.

This improved the disconnected error for other updates but MQTT updates remain unreliable via BLE.

c-:

RichieFrame, 2024-12-18:
the disconnect after setting most settings is because the device reboots to apply the setting

ctmalan, 2024-12-18:
May I suggest that it be acknowledged before rebooting?  For some settings
this is successful.

Maybe the race condition happens where the acknowledge message is longer
than anticipated.
--
c-: