meshtastic/Meshtastic-Android #1483: [Feature Request]: Stop notification sounds on boards if you are in DND on smartphone

Dtrieb, 2024-12-18:
### Tell us your idea.

I'd like the app to stop notification sounds on boards if you are in DND on Smartphone.

P.e. if you set your phone on DND and forgot that the tracker plays it's tune, it very annoying in a meeting.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

todd2982, 2025-03-27:
This would involve rebooting the node each time the phone's DND setting was toggled.  This would cause the T1000e to play the boot-up tone each time.  Is this something you would still like somebody to look into?

jamesarich, 2025-03-31:
I don't think it makes sense to tie these things together, they are both individually settable depending on where/how you'd like to be notified.

closed 2025-03-31 by jamesarich