meshtastic/Meshtastic-Android #1484: [Bug]: Blocked/Ignored nodes not removed from map

b8b8, 2024-12-18:
### What happened?

I have been testing the new ignored node feature with the app and noticed that when you ignore a node it does not remove it from the map. I understand and agree that the ignored node should be still on the node list crossed out so you can un-ignore it at a later date if you so choose, however i believe all positions shown on the map should also be removed/hidden. My specific use case is blocking a node with an inverted longitude value causing it to always show up in China, while i know it is only a few km away. This "faraway" node also causes the map to zoom wayyy out when loading.

### App Version

2.5.14

### Phone

Pixel 7a

### Device

SEEED T1000E

### Firmware

2.5.16

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

SimbimChimbetov, 2024-12-18:
Try Block and forget the Node. If you want unblock later, you can do it in LoRa Menu.

b8b8, 2024-12-18:
if you click ignore node then the "remove" is grayed out. If i just click "remove" node then it will come back when the next packet is received from that node. You still want "ignored" nodes to show up on the list but not the map so you can un-ignore them in the future.

PR #1676 (2025-03-17): fix #1484: Filter out ignored nodes from map
- Created a new `filteredNodeList` `StateFlow` in `UIState.kt` that filters out nodes marked as ignored.
- Updated `MapFragment.kt` to use `filteredNodeList` instead of `nodeList`, ensuring that ignored nodes are not displayed on the map.

fixes #1484 

https://github.com/user-attachments/assets/eab5b07a-9f65-49fe-8a64-421e5b0586fb
commit 8e5accd5 (2025-03-17): fix #1484: Filter out ignored nodes from map (#1676)

* fix #1484: Filter out ignored nodes from map

- Created a new `filteredNodeList` `StateFlow` in `UIState.kt` that filters out nodes marked as ignored.
- Updated `MapFragment.kt` to use `filteredNodeList` instead of `nodeList`, ensuring that ignored nodes are not displayed on the map.

* Refactor: Remove ExperimentalCoroutinesApi opt-in and add it to compiler args

- Removes the `@OptIn(ExperimentalCoroutinesApi::class)` annotation from multiple files.
- Adds `-opt-in=kotlinx.coroutines.ExperimentalCoroutinesApi` to the `freeCompilerArgs`  [... cut]
closed 2025-03-17 by jamesarich