meshtastic/Meshtastic-Android #1492: [Bug]: Message input auto-capitlization

madeofstown, 2024-12-20:
### Contact Details

@madeofstown

### What happened?

In previous app versions when typing a message, the first letter of each sentence was automatically capitalized. This behavior is expected and is the default for almost all text entry boxes. A few versions ago (not sure exactly when) that changed and first letters are no longer capitalized automatically.

### App Version

v2.5.13, v2.5.14, v2.5.15

### Phone

Pixel 7a, GrapheneOS

### Device

T1000e, Linux Native Portduino

### Firmware

V2.5.15, v2.5.16

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

madeofstown, 2024-12-20:
[image]
It's not a keyboard settings issue. Message input for SMS messages automatically capitalizes as expected

andrekir, 2024-12-22:
Good catch, this will be fixed in the next release (fixed in e924f274f150487a13e386a3774729b4edca93bf).

closed 2024-12-22 by andrekir