meshtastic/Meshtastic-Android #1496: [Feature Request]: Clear old data button

GMart, 2024-12-21:
### Contact Details

Discord Droidbot2

### Tell us your idea.

I'd like the app to have a "Clear old data", "Delete old data" or maybe "Clear bad data" button.
It could:
Clear old traceroutes or Clear traceroutes that have no response since those don't have any data.
Clear data in all metrics logs older than X weeks
Clear nodes from the attached device that have no full nodeinfo, would be good for traveling since you probably won't get the full node data anyway.

Kinda like the "Delete browsing data" button in browsers.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GMart, 2025-03-07:
This would be different than clearing the NodeDB. 
The proposed button would not clear nodes in the DB that name NodeInfo or location data. It would only clear nodes with no NodeInfo, along with the traceroutes associated with them. This would help clean up the Node list.
Maybe similar to how MeshSense can clear nodes last seen older than X days.

todd2982, 2025-03-27:
Click the three dots setting icon in the top right, select "Debug Panel".  Here you can select the "Clear" button.

GMart, 2025-04-22:
> Click the three dots setting icon in the top right, select "Debug Panel". Here you can select the "Clear" button.

That clears all data. What if I want to keep recent data, or I have moved locations and only want recent nodes?

My meshtastic App is using over 400MB of data on my phone. How much of that is old signal, telemetry, and environmental data from Nodes that I'm not connected to? I don't see a way to view data for nodes that I don't currently have in my active node's NodeDB.

jose-d, 2025-05-25:
...or a configurable TTL? E.g., disregard nodes that haven't been seen in the last {n} days?

closed 2025-07-03 by github-actions[bot]