meshtastic/Meshtastic-Android #1501: [Bug]: Configuration Import/Export options not always visible in app

rmeineke, 2024-12-22:
### Contact Details

rmeineke@gmail.com

### What happened?

The option for importing and exporting configuration files is not always visible. I can see it initially when I go to the Remote Administration tab. But if I choose any other option, e.g. Pax Counter, when I return to the remote administration tab the import/export options are not visible. If I go back one more page and then select Remote Administration again, then they become visible again.

This discussion thread: 
https://github.com/orgs/meshtastic/discussions/143
has some more background, screenshots, etc.

### App Version

First noticed this in version 2.5.12. Prior APKs downloaded and tested do not have this behavior. Latest 2.5.15alpha has the problematic behavior as well.

### Phone

NUU phone, model S6512L, Android 13. I also see this behavior on an old Axon7 phone running Android 8.

### Device

Behavior is the same for my Heltec v3.1 devices and my RAK4631 devices.

### Firmware

A variety: one Heltec device tested is running 2.5.8, one RAK4631 is running 2.5.11.

### Relevant log output

```shell
Happy to add this info but you'll need to point me in the right direction.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-12-23:
Thanks for reporting. This is a known issue and will be fixed when #1460 is merged.

andrekir, 2025-01-09:
this is fixed in release version 2.5.16