meshtastic/Meshtastic-Android #1504: [Bug]: T1000E memory wipes on bluetooth reconnect

b8b8, 2024-12-23:
### What happened?

Im not sure if this is specific to android so might actually be a firmware issue.

I was running firmware 2.5.16 on the T1000E and walked into and out of bluetooth range a few times (3) of my android phone and the phone wouldnt reconnect. The T1000E completely locked up. I could get it into DFU mode and tried to reflash the firmware but that did not work. I used the flash erase file then was able to restore the firmware. Fortunately i did not need to reload the bootloader. 

This issue has happened to me maybe 4 times since the T1000E was released so what ive been doing is manually disconnecting from the device before getting out of bluetooth range. Whenever i stop manually disconnecting from the device and let my normal comings and goings disconnect from it naturally, i experience the file system issue within about 2 days.  

### App Version

2.5.15

### Phone

Pixel 7a

### Device

SEEED T1000E

### Firmware

2.5.16

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-23:
I have tried to cause this on iOS for months and have not had any success, when we first did a bunch of BLE work in the firmware (which was initially incompatible with CoreBluetooth on iOS) I copied a bunch of settings from android and then at some point we added the same disconnect / timeout timings as CoreBluetooth to android. I think at one point android was doing something on the filesystem when connecting / disconnecting but I feel like that was removed @thebentern do you remember any additional details?

ravenium, 2025-01-23:
I had something similar happen - it looked like an lfs corruption of some sort. I saw a possible fix in .18 - I haven't tested much, but so far so good...

todd2982, 2025-03-27:
Is this still happening on firmware 2.5.20 or newer?  I suggest closing this ticket based on the comments above if there is no response.

ravenium, 2025-03-27:
It appears that 2.5.20 made it so that it doesn't brick - it merely resets to defaults (i keep a backup config for this purpose).  I have not had this problem at all since 2.6.x, and based on the changelog I have reasonable confidence that this is fixed.

closed 2025-03-27 by b8b8