meshtastic/Meshtastic-Android #1509: [Bug]: Unable to change channel index with drag and drop

IlyaMZP, 2024-12-28:
### Contact Details

ilya@mango.vg

### What happened?

When dragging a channel to change its id, it gets replaced in the middle of dragging. This makes changing channel ID impossible.

https://github.com/user-attachments/assets/e52de2e6-ae9a-4b1b-9b75-6d33a2fdb343

### App Version

2.5.15

### Phone

OnePlus 8

### Device

Heltec V3

### Firmware

2.5.15.79da2365

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

IlyaMZP, 2024-12-28:
It seems to work in the "Radio configuration" menu under "Channels" category.

RyanWor, 2025-01-09:
Same issue on Pixel 9 Pro Android 15 app version 2.5.15. Force killing the app does not resolve the issue for me, but the workaround to reorder channels from the configuration menu rather than the channels tab in the main UI did work for me. Thanks!

todd2982, 2025-03-24:
I feel like changing channels at this screen should be disabled.  Does this ignore the "Managed Mode" security option?