meshtastic/Meshtastic-Android #1510: [Bug]: High node numbers result in delete messages and constant new node

DooMMasteR, 2024-12-28:
### What happened?

I am and many others are currently at 38C3 and there is a very buys meshtastic mesh on and around the event, my node alone saw ~48000 packages in a day.  
With more than ~80-100 nodes, both the App and the node develop issues.  
The first one is that they are constantly notifying of new nodes joining the mesh, the node (T-Echo) is limited to 80 nodes, but the app should be able to track more, it however seems to fail at it and constantly notifies me about new nodes I have 100% already seen in the past.  
But even worse, the app lost multiple private chats with other nodes after forgetting the node. 
Even my own relay (another T-Echo) and it's messages got lost this way.  

### App Version

2.5.15

### Phone

Pixel 8 Pro

### Device

T-Echo

### Firmware

2.5.15

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

infered5, 2025-01-05:
Can confirm this happens to me too. Using a RAK4631 and Galaxy Flip 3 (Android 14).
Meshtastic app 2.5.15 (Play store with Beta Tester access). Seems to happen around the high 80s of nodes in memory.

jpryor223, 2025-01-11:
I am seeing the same issue as well, any nodes over 80 or 100 depending on which device are not loaded on reconnect. In particular I see nodes that are not encrypted (older firmware) mostly missing that were active in recent hours and all need to request info and position again. Other unset and unrecognized nodes are also not reloaded. The app node DB continues to have history for older nodes so once the node appears again I can see past logs, just takes time for info and position to update again like finding a brand new mode.

jpryor223, 2025-01-11:
Also should note I am not seeing this issue from Apple app, that is maintaining more than >100 nodes on reconnect.

todd2982, 2025-02-24:
Version 2.5.17 now adds favorite nodes.  A favorite will not be removed from the nodedb when it becomes too large.

jpryor223, 2025-03-01:
What is the definition of nodedb becoming 'too large'?  Is that app and/or node device specific size?  If that is the 80 or 100 node limit based on the particular device/firmware?  I have been using 2.5.17 beta on Android and have favorited a number of nodes.  The app continues to purge nodes down to 80 after reconnect, and some nodes that I had favorited continue to get removed.  I typically get about 20 new node notifications a day as the app builds back up to 100-120 active nodes, and most of these are nodes seen before.

todd2982, 2025-03-07:
Yes, the nodedb is limited to 80 or 100 nodes for NRF52 and ESP32 type radios respectively.  If you are seeing favorite nodes being removed, then this is something that should be submitted as a separate issue to the firmware repo as this would be seen for Android or iOS.

I suggest upgrading to the latest beta (2.5.20 as of writting), and see if you still have issues with favorite nodes being removed before submitting a new issue to firmware for that problem.

garthvh, 2025-03-07:
iOS does not prune messages or nodes from the app, so these details must be Android related

jamesarich, 2025-03-07:
To @garthvh 's point - currently node data is not stored app-side on Android and relies on the device to provide the nodedb on connection. The way the Android app behaves tends to fall back to the device as the source of truth in a lot of cases.

jamesarich, 2025-05-01:
@DooMMasteR as of the latest alpha builds (since 2.5.22-alpha) the app now retains data between connections to the same device. Please test with a recent build, we can re-visit if your issue is not resolved.

closed 2025-05-01 by jamesarich

jpryor223, 2025-05-01:
Thank you, confirm seeing this issue resolved in 2.5.22 alpha, all nodes are no longer being cleared on every reconnect, and that node history remains again in the app.