meshtastic/Meshtastic-Android #1528: [Bug]: S&F on the public channel

jranma, 2025-01-07:
### What happened?

Android app version 2.5.15
Node1 firmware : 2.5.15.79 Node 1 is a S&F server. 

When I send a **direct message** to node 1 from another node to retrieve S&F messages using the “SF” command, I receive the reply: “S&F not permitted on the public chanel”. 

However, I'm not on the public channel (Longfast, or any other channel), I'm sending via a direct message. 

### App Version

2.5.15

### Phone

Android 14

### Device

Lilygo T3S3 e-ink

### Firmware

2.5.15.79

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2025-01-07:
Your primary channel and the frequency slot is the public channel

jranma, 2025-01-07:
Sorry, I don't get it. 
to query the SF server, you have to use a direct message, don't you? 
So what's the meaning of this error message? Or to put it another way, how can we query the SF server if we can't do it via a direct message?
Best

caveman99, 2025-01-07:
The S&F server will refuse to service requests when it has the default channel key configured.

jranma, 2025-01-08:
> The S&F server will refuse to service requests when it has the default channel key configured.

Okay, I think the aim is to avoid saturating the channel with S&F messages. 
But I still don't understand this, since communications with the S&F server are made via direct messages, which are invisible to other users. What difference does it make, in terms of radio frequency use, whether or not the S&F server is set up with the main channel? If it's set to another channel, the radio frequency will still be used, right ? Am I missing something here?

garthvh, 2025-01-08:
> Okay, I think the aim is to avoid saturating the channel with S&F messages. But I still don't understand this, since communications with the S&F server are made via direct messages, which are [...]

Direct messages are not isolated as you seem to be assuming they are here, they are creating channel utilization via the primary channel or a dm.

jranma, 2025-01-10:
ok thanks for the info. 
makes sense now. 
I guess you can close this issue.

mangelajo, 2025-06-11:
I don't understand why this is closed, I am still trying to understand what is the issue.

If I am writing a message to the SF server, using the legacy "SF" command or anything, I still get the error.