meshtastic/Meshtastic-Android #1546: [Bug]: Remote Admin Shutdown Does Not Execute

IcyMidnight, 2025-01-13:
### Category

Other

### Hardware

T-Echo, Station G2

### Firmware Version

2.5.15.79da236

### Description

Using remote admin, I am able to change settings and reboot the other device reliably. Sending the shutdown command never has any effect, even if the command is confirmed delivered.

jsoberg, 2025-02-09:
The shutdown command from my Android to my T-Echo works as expected, firmware 2.5.20. Is it possible there was a fix in the firmware between 2.5.15 and 2.5.20? I did a quick search and only found https://github.com/meshtastic/firmware/pull/5859, which appears to be unrelated

Robert-0410, 2025-02-19:
Android Client Version: 2.5.17

Confirmed problem on firmware 2.5.15.79da236.

Flashed 2.5.20.4c97351 and the command works as expected.

jamesarich, 2025-02-19:
Working as expected on newer firmware

closed 2025-02-19 by jamesarich