meshtastic/Meshtastic-Android #1547: [Bug]: No unit shown next to pressure value

b8b8, 2025-01-14:
### What happened?

Self explanatory :).
[image]

### App Version

2.5.16

### Phone

Pixel7a

### Device

N/A

### Firmware

N/A

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Kealper, 2025-01-15:
Hmm, the unit that's displayed is hPa, hectopascals, but that doesn't have a very good short symbol. I could try a very small 'hPa' to the right of it as a mock up later tonight to see if that might look okay.

b8b8, 2025-01-15:
All the unit symbols could be smaller if required to get it to match better, or maybe just lower the font by one point for everything.

Kealper, 2025-01-15:
Yeah, I'm thinking just having `hPa` in there would fit just fine. I'm also considering `MΩ` for the Gas resistance box.

b8b8, 2025-01-15:
I think also radiation should be uR/h. But someone else please confirm.

Firecul, 2025-01-30:
> I think also radiation should be uR/h. But someone else please confirm.

IIRC that is the unit that should be used yeah. And I definitely think that radiation needs units because you can get vastly different levels of exposure from something that might not look that different if all you are seeing is a number.

b8b8, 2025-01-30:
We already have units for radiation. It just dropped the /h part. 

[image]

Kealper, 2025-02-14:
Got busy and only just now go around to adding those units, oof!

Tested with a bit of mock data and it displays fine, there's already code in the info-card generator that will make the size of the text for the value smaller if it would be too long, so things like `1016 hPa` or `265.0 µR/h` just render as slightly smaller text than things like `23°C`. Convenient!

PR #1583 (2025-02-16): fix: Add missing sensor value units to Node Details cards
This PR adds the units for sensor values that didn't have them. Fixes #1547
closed 2025-02-16 by jamesarich