meshtastic/Meshtastic-Android #1548: [Bug]: Wrong unit used for position precision

draen, 2025-01-14:
### Contact Details

draen94 (discord)

### What happened?

Position precision slider shows imperial units, any other value uses metric units.

Phone:
- language set to English (United Stated).

Meshtastic app:
- Radio Configuration > Display Config > Display units set to metric

[image]

### App Version

2.5.15

### Phone

Motorola One, Android 10

### Device

Heltec Wireless Stick Lite v3

### Firmware

2.5.15

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-01-14:
You need to change your language settings to a metric country like Canada.

draen, 2025-01-14:
That helped, thanks. Is this an expected behaviour? I haven't noticed this "problem" in any other app.

b8b8, 2025-01-14:
Units are derived from you phone location or region settings. The display config you changed in the app is only for the meshtastic device LCD.