meshtastic/Meshtastic-Android #1552: [Bug]: Android App never shows the PAX counter value.

Miaoucat0, 2025-01-19:
### Contact Details

kenji.acso@gmail.com

### What happened?

the Android App never shows the PAX counter node count.
ive been trying various things for months now and it never worked once.
it seems the PAX node is working as expected, but the Android APP doesnt show the number.
i can see the grafts for signal strength and such, but nowhere is the PAX info.

tried with the receiving node being Heltec V3 and TTGO 1.6.1 model.
Android Meshtastic app:  2.5.16
both nodes are using the default LongFast channel as primary

[image]

### App Version

2.5.16

### Phone

Samsung S22

### Device

heltec V3, TTGO 1.6.1, Heltec Wireless Paper,

### Firmware

2.5.18 and <2.5.13 i think. both fairly recent version

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-01-19:
You turned off bluetooth as well on the pax node?

Donnie650, 2025-01-19:
I can verify this is true. I setup a python script to send me a message when updated because android app never updates or if it does is over written on next update and you never see it again.
Please note at the top of pic. My message arrives but no up

[image]

date on android

Miaoucat0, 2025-01-20:
> You turned off bluetooth as well on the pax node?
> 
> [image]

Bluetooth and Wifi are both off on the Paxcounting node. (wireless paper)

Miaoucat0, 2025-01-20:
**[UPDATE] new information**

i think the problem is indeed that the app doesnt keep the last PAX count on the screen of the app.
after a short while, if the app is minimised or opened, if there is somekind of screen refresh the numbers dissapear.

look at this:
first time ever i see the PAX count on the screen, i check the SMS messages on my phone and come back- its gone.
[image]

Donnie650, 2025-01-20:
I have seen that happen only twice. It was there then it was not. and it would never display again. It's not firmware it's the app that's not handling it right for sure.

b8b8, 2025-01-21:
This explains the varied results we were all getting. May have to get an automatic update kinda like the node "last heard" time did.

Miaoucat0, 2025-01-21:
i dont know anything about the programing so its in your hands. i am happy i could have been of help.

i think adding the logging/graph capabilities to the PAX counting information might be a good idea no?
many people would greatly appreciate being able to monitor the PAX count across time. similar to the SNR, Battery level and such other information. anybody wanting to keep that information private could simply put the PAX node on a secret channel so that other cannot see it. i see this as only benefit of having this feature.

anyway, how does this work? should i keep this thread open so other people can comment their findings and progress for this problem? we have found the key element, but i dont think this place is where the programmers are hanging out no?

b8b8, 2025-01-21:
@andrekir is the main Android guy, and @Robert-0410 has been doing the new android graphs.

Robert-0410, 2025-01-21:
#### PAX Graph
Potentially in the future I can pick this up, but it depends on demand for the feature and making sure it aligns with the bigger picture.

Donnie650, 2025-01-22:
It's kind of hard to do if the feature is not working people just stop trying to use it. Get it working and the log will be requested for sure.

MagooYFZ, 2025-01-30:
For some reason its not working and they don't seem interested to find and fix this bug.

I have given up on this feature.

b8b8, 2025-01-30:
I'm sure it's just not a high priority for the devs right now and they have don't have a personal use for it to motivate the change. You can always code it yourself, but this is volunteer so devs work in the areas they want and have use for. It will probably get updated eventually.

Donnie650, 2025-01-30:
Make that two of us who are disappointed. I have tied for a full year too get this fixed. This project was fun, now...... not so much now.

b8b8, 2025-01-30:
The entire meshtastic project is now longer fun because pax doesn't work as expected? Other projects use pax already.

Donnie650, 2025-01-30:
Its the full year of trying to get a bug fixed, that some of us could use but are just ignored. Thats what makes it not fun. Its how this project has been for a while now. core group gets things they want the rest of us wait and hope our bug will be fixed next. A year later and still nothing has been done, that's the disappointment.

Miaoucat0, 2025-01-31:
the worst i think is the number of times this issue was brought up and dismissed. searching around this morning i saw plenty of places were people complained rightfully that this feature wasnt working and ended up quitting the subject. people woudnt have cared about this so much if this feature wasnt on the app broken, for such a long time. 

it is totally normal to keep things like this in the private spheres of the developers until it is ready to be released. the problem is that people like me saw this on the app, and information guides like : https://meshtastic.org/docs/configuration/module/paxcounter/  and expected it to work for a long time in vain. if only there had been acknowledgment of this buggy stage. like a simple warning on the wiki "this feature might not work on Android". i am pretty sure i even asked on the Discord one time for this and it got dismissed too.
 
things like the new graphical interface are rightfully kept from being advertised to the general public until their proper release. this is exactly to avoid the problem we have right now with PAX counter. also it is not because this feature isnt used by some that it isnt extremely useful for others. it is why there are so many buttons on a keyboard, they all have their uses and not having them would be a great loss.

b8b8, 2025-01-31:
### 

> _**personally i am planing on installing this at my car to notify me thru the mesh when somebody else has arrived at the parking spot while i am away deep in the forest.**_

That's actually a pretty interesting use-case.

todd2982, 2025-03-09:
Can this be closed after #1635?

jamesarich, 2025-03-09:
fixed in #1635 0

closed 2025-03-09 by jamesarich