meshtastic/Meshtastic-Android #1559: [Feature Request]: Add rain sensor data to the Environment Symbol Overview

MALAONE1, 2025-01-26:
### Contact Details

manuel.lausmann@web.de

### Tell us your idea.

Hi, it would be great if two new environment symbols and data were added to the DFrobot rain sensor that @caveman99  added to the firmware for rain from the last 24 hours and the last hour. Then you have a nice overview of  the other sensor data. Thanks your Work

[image]
[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-02-16:
this will be available in the protos to begin surfacing to the ui in #1589

MALAONE1, 2025-03-04:
The rain is not yet visible🥲

[image]

closed 2025-05-24 by github-actions[bot]