meshtastic/Meshtastic-Android #1562: [Feature Request]: Allow ip access when not connected to wifi

hcker2000, 2025-01-28:
### Contact Details

hcker2000@gmail.com

### Tell us your idea.

It would be nice if the IP option was allowed as a node connection even when off wifi. Anyone using a VPN to their house could access base stations while out and about 

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

madeofstown, 2025-01-29:
With the current implementation you can already connect to the node (in home network) while using a VPN to your home network. The issue is that the IP will not show up in your node list. You just need to remember the IP and type it in. 

The reason for no AP on the nodes is addressed every few weeks here by the admin. It's not a good user experience (slow and unreliable, plus the lack of an internet connection confuses users)

hcker2000, 2025-01-30:
Interesting I have tried that but it never connects. Is it using tcp or Udp? I can try and dig in more but I know I can access everything else on my network with no issue.

This is on the android client.

[image]

madeofstown, 2025-01-30:
What sort of VPN are you using? I've only tested with standard wireguard so I'm not sure about any other VPN software. I'm pretty sure it's the HTTP API that Meshtastic is using for IP connections so that would be TCP.

hcker2000, 2025-01-30:
It is a wireguard vpn, typical travailing salesman type setup. Sounds like this is my issue and not the app.

closed 2025-01-30 by hcker2000