meshtastic/Meshtastic-Android #1567: [Feature Request]: Allow using a different TCP port than port 80 when specifying module IP address

fernandosanchezjr, 2025-02-01:
### Contact Details

fernando.sanchez.jr@gmail.com

### Tell us your idea.

I'd like the app to allow me to specify a port other than TCP port 80 when connecting to a meshtastic device via HTTP. ❤ 

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PhantomSage, 2025-02-15:
Clarification, the app does not use port 80, it uses the 'binary' API on port 4403.
It's possible to forward this port from a "normal" machine to the node if that helps. 
It took a bit of digging, since it's not evident in the documentation.

I agree with @fernandosanchezjr that the ability to specify port would be preferred, then I could have multiple nodes on a single IP address acting as TCP proxy.

fernandosanchezjr, 2025-02-15:
> Clarification, the app does not use port 80, it uses the 'binary' API on port 4403.
> It's possible to forward this port from a "normal" machine to the node if that helps.
> It took a bit of digging, [...]

Thank you for the clarification. My use case is precisely the TCP proxy situation you've described.

commit e90aa6c5 (2025-05-19): fix #1567: Allow using a different TCP port than default 4403 (#1861)
closed 2025-05-19 by jamesarich