meshtastic/Meshtastic-Android #1572: [Bug]: UI: Can send a message towards an out-of-range user.

shortwavesurfer2009, 2025-02-04:
### Contact Details

this issue comments

### What happened?

I see a node in my list that is 5 hops away and yet the app still allows me to click direct message and send them a message as if they would actually receive it.

My radio is set at the default 3 hops.

Either the app needs to not show devices that are more than the set number of hops away or messaging, request user info, traceroute and other functionality needs to be disabled for those nodes.

### App Version

2.5.16

### Phone

OnePlus Nord N200

### Device

SeeedStudio SenseCAP T1000-E

### Firmware

2.5.18

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

GUVWAF, 2025-02-04:
The "hops away" count is from a packet from that node towards you. This does not necessarily mean that the route from you towards that node needs the same amount of hops.

shortwavesurfer2009, 2025-02-04:
Ah, okay

closed 2025-02-04 by shortwavesurfer2009