meshtastic/Meshtastic-Android #1576: [Bug]: Fixed Position gets overwritten

ZuluZuluNL, 2025-02-09:
### What happened?

I try to set new location settings manually. After saving I see that these values are stored and the new position is shown on the map. However after a couple of minutes the new settings are overwritten by the old (initial) settings.

As I have seen other users also raised bugs about this setting in the past I have tried to update the fixed location settings with other settings. For example switching on and off GPS mode etc. 
Also tried to remove fixed position. Save. Reboot. Re-enter the new settings. Save. Reboot.
Even after performing a full factory reset the settings get overwritten again.

### App Version

2.5.16

### Phone

Samsung Android 14

### Device

Heltec V3

### Firmware

v2.5.18.89ebafc

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ZuluZuluNL, 2025-02-09:
Found the cause myself. Seems that the new location_precision setting is messing up the manual setting. Although I support this new feature, I don't think putting this parameter under the channel settings is very intuitive.

closed 2025-02-09 by ZuluZuluNL

ianp5a, 2025-03-23:
I think this is a usability problem that needs fixing. I had the same issue in 2.5.19 only solved if Channels has Position Disabled. Other new users will have this problem. An INFO button could help as a quick workaround.