meshtastic/Meshtastic-Android #1577: [Feature Request]: Triggering NodeInfo send from application

phaseloop, 2025-02-09:
### Tell us your idea.

Meshtastic firmware allows to trigger sending NodeInfo packet (out of schedule) by fast pressing button twice. 

It is convenient for testing radio propagation and connectivity to check online which MQTT nodes received our packet (without need to wait for scheduled NodeInfo which may happen 2-3h later).

It would be great to be able to trigger that from the application.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-02-09:
Actual nodeinfo can be sent as frequently as every 3 min should your node (or another node) get a packet from a node not yet on the node list. The 3 hour minimum for nodeinfo is only for automatic broadcast interval, not if the node is explicitly asked to produce said nodeinfo to identify an unknown node.

madeofstown, 2025-02-11:
I think this is a great idea. I would also like to see a way to send position data to the mesh triggered from the app (instead of requesting).

b8b8, 2025-02-12:
> I think this is a great idea. I would also like to see a way to send position data to the mesh triggered from the app (instead of requesting).

Request position is actually "exchange position" its named as such in the iOS app but it does the same thing in both apps.

madeofstown, 2025-02-12:
> Request user info works in the android app.

Does "Request user info" function similarly to exchange position? If I request user info am I sending my user info out? This is the function that OP wants. Transmit user info, initiated from the app.

garthvh, 2025-02-12:
The double click functionality sensor a position packet and falls back to a node info if a position is not available.  The node info is sent out when necessary for DM's automatically, really no reason to impatiently request it manually.

madeofstown, 2025-02-13:
Garthvh *really* hates node info broadcasts, lol.

It can be done from the physical device, it can be done from the python command. I think people were just hoping it might get implemented in the apps too. I guess we have our answer.

Kealper, 2025-02-16:
`Request user info` works the same way as `Request position` in the app, it actually send out your data (user info packet or position packet) over the primary channel with the specified node as the target with a flag on it that basically tells the node at the other end "here's my data, please reply back with yours".

closed 2025-05-24 by github-actions[bot]

Terramoto, 2025-09-04:
I think it would be nice to have a button to manually broadcast a nodeinfo in CLIENT_HIDDEN mode with a 5 minute timer.
In an area with a good coverage and populated it would be more beneficial than the CLIENT_MUTE, i think (could also be wrong).