meshtastic/Meshtastic-Android #1592: [Bug]: No channel while trying to send DM

danditomaso, 2025-02-17:
### Contact Details

Dandcodes

### What happened?

I have a friend I DM with frequently, my application was saying I was connected to my Heltec Meshtastic device. Anytime I send a DM it returns saying "no channel" which isn't a very helpful error, and doesnt describe how I can fix the potential issue. 

### App Version

2.5.16

### Phone

Pixel 6 Android 15

### Device

Heltec v3

### Firmware

2.5.20

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

resunltd, 2025-02-17:
I have a similar issue using Station G2.
Attempting to send a DM to radios that I have been daily messaging has stopped working. Android APP shows instant cloud with slash indication upon SEND. Both ends showing Green Lock. Other radios in mesh can be DM'd as usual without issue from the G2.

Also tried deleting the nodes to refresh the key.

Any ideas?

garthvh, 2025-02-17:
This comes directly from the firmware