meshtastic/Meshtastic-Android #1593: Android app is harder to close than vim

fifieldt, 2025-02-18:
### Tell us your idea.

Per https://github.com/meshtastic/Meshtastic-Android/issues/186 , there is no way to stop the android app other than using "Force Stop" in system settings.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

fifieldt, 2025-02-18:
In [discord](https://discord.com/channels/867578229534359593/931765413350613052/1341250668354207754), we also had a bit of a discussion with the idea to try and figure out why our users are haing this experience with the meshtastic app in particular.

Robert-0410, 2025-02-18:
> Per [#186](https://github.com/meshtastic/Meshtastic-Android/issues/186) , [#303](https://github.com/meshtastic/Meshtastic-Android/issues/303) , [#422](https://github.com/meshtastic/Meshtastic-Android/issues/422) and [#527](https://github.com/meshtastic/Meshtastic-Android/issues/527) there is no way to stop the android app other than using "Force Stop" in system settings.
> 
> Given this has been requested 4 times (now 5) by the community, this should likely be addressed.

Is this even true given that the user can select "None" --> close the app and the OS will stop the service?

fifieldt, 2025-02-18:
My conclusion from the chat linked above is the root cause of continued user requests for an exit button is the current exit flow is not sufficiently smooth/obvious. My _guess_ is that people expect the app to close fully on app-menu swipe, without having to take extra action within the app. Whether this is best fixed with modifications to the swipe behaviour or changes to the node selection UI I would leave in more capable people's hands!

Kealper, 2025-02-18:
Personally I never minded the exit flow being disconnect from node > swipe app away to get it to be completely closed as it seemed to make sense to me, but as a napkin-math kind of idea: Perhaps a checkbox that's checked by default right under the "Provide phone location to mesh" checkbox on the connection tab that says something to the effect of "Keep app running in background when connected to mesh" that, when unchecked, causes the app to disconnect from the current device when swiped away from the open app list. Basically just automating the existing way of exiting while keeping the current behavior the default.

Robert-0410, 2025-02-18:
> Personally I never minded the exit flow being disconnect from node > swipe app away to get it to be completely closed as it seemed to make sense to me, but as a napkin-math kind of idea: Perhaps a [...]

Yea, this idea is on the right track. It would preserve the default behavior that's used by many and gives the option for those that don't need it. Not sure on the placement of the preference, I think it would be important to maintain a separation between radio and client configuration/preferences. I'm not really sure how that looks like right now.

b8b8, 2025-02-18:
Yea the nodes are kinda designed to be connected all the time. There should be some way to keep it running in the background if desired (whether a toggle or check box) so swiping doesn't kill app accidentally. Like an add blocker app or VPN, or even Garmin smartwatch app, there should be a way to keep a persistent connection if desired.