meshtastic/Meshtastic-Android #1603: [Bug]: Older versions of Android showing strange UI when focusing text fields in settings

jsoberg, 2025-02-22:
### What happened?

User on [Discord](https://discord.com/channels/867578229534359593/871539863307055134/1342909101713260687) noticed that on their Samsung S9 (Android 10) after updating to 2.5.27, the settings UI seems to squish and show the main UI behind it when focusing on a text field. I was able to repro this on my Samsung A10e (Android 9): 

<video src="https://github.com/user-attachments/assets/3a6345cc-39f5-405a-8ede-71322420bf38" width="300"></video>

Note that the content is scrollable.

Also note that I cannot repro this on my Samsung S23 Ultra (Android 15).

### App Version

Latest master: 3e74373617ed483411d0a23f5d71d0c78220efeb

### Phone

Samsung A10e, Android 9

Samsung S9, Android 10: https://discord.com/channels/867578229534359593/871539863307055134/1342909101713260687

### Device

N/A - App issue

### Firmware

N/A - App issue

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jsoberg, 2025-02-22:
Ran a quick check - rolling back to Compose BOM `2024.12.01` (our previously used version) doesn't fix this.

Mictronics, 2025-02-26:
Confirmed on Samsung SM-T813, Android 9, Lineage OS

mrekin, 2025-02-28:
There are lot of UI "artifacts" on my Oneplus 7T (android 10) - mostly with input fields ( include highly used messages inputs), but with other elements too - near 70% transparent device role select list as example..

jsoberg, 2025-02-28:
@mrekin the issues you're describing should be fixed with https://github.com/meshtastic/Meshtastic-Android/pull/1605 which will be included in the next update 🤞

jamesarich, 2025-02-28:
fixed by #1605

mrekin, 2025-03-07:
release 2.5.18 - fixed semitransparent dropdowns, but bug with inputs still here.

PR #1771 (2025-04-21): fix: Remove imePadding modifier to internal scaffold box
fixes #1603
fixes #1767

`BaseScaffold` is applying `imePadding()` to an internal `Box` on all screen-level Composables. This is evidently unnecessary with our setup and applies double keyboard padding for older devices.

On my Samsung A10e, Android 9:
| Before | After |
|------|-----|
| <video src="https://github.com/user-attachments/assets/021e275c-0628-46b2-aa73-639ec5a4ed4c" width="300"></video> | <video src="https://github.com/user-attachments/assets/57737d25-4a3f-477c-be53-e1d7956e5a83" width="300"></video> |

Also tested on my Sam [... cut]
closed 2025-04-21 by jamesarich