meshtastic/Meshtastic-Android #1604: [Bug]: Light theme has color issues from surface overrides

jsoberg, 2025-02-22:
### Contact Details

`myheartpills` on discord

### What happened?

While using light mode, I noticed that all cards (and some other UI elements) in the app have a theming issue showing strange outlines/colors:

Cards:
[image]

Switches (semi-transparent):
[image]

This is likely due to us setting a surface override in our theme that is semi-transparent, meaning there will be some issue when the surface overlays background (which isn't overridden). We can't reasonable set a background color with non-FF alpha either, since this will mean backstack entries will show below.

### App Version

latest master [3e74373617ed483411d0a23f5d71d0c78220efeb]

### Phone

Samsung S23 Ultra, Samsung A10e

### Device

N/A - app theme issue

### Firmware

N/A - app theme issue

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jsoberg, 2025-02-22:
These are some options we have:

1. Set an `AlmostWhite` value that is fully opaque. This as an example is `0xFFEFEFEF`:

[image]

Note that this can look "more normal" if we decrease the elevation on the cards. Our current value of 4dp is a bit large since there is only 2dp of space between each card.

2. Stop overriding surface and onSurface completely, just using default material values:
[image]

jamesarich, 2025-02-22:
I'm on board for option 2 - let's get it back to standard Material

PR #1605 (2025-02-22): Fixes #1604: Stop overriding surface/onSurface in theme
Fixes #1604: Stop overriding `surface` and `onSurface` in our `AppTheme` to remove issues from semi-transparency.

| Before | After |
| ------ | ------ |
| [image]  | [image]  |
| [image]  | [image]  |
commit ef64df35 (2025-02-22): fix: Stop overriding surface/onSurface in theme (#1605)

Fixes #1604
* Stop overriding surface/onSurface in theme

* Wrap preview in AppTheme to get a better representation of what it'll look like on-device
closed 2025-02-22 by jamesarich