meshtastic/Meshtastic-Android #1611: [Feature Request]: ROUTER "are you certain?"

mahlonsmith, 2025-02-25:
### Contact Details

mahlon@martini.nu

### Tell us your idea.

I am noticing an uptick in new nodes in my area (awesome!), that are clearly in lower elevation areas and are set to ROUTER (not awesome!)

I'm afraid that these nodes are 'poisoning the well', and while I've tried (unsuccessfully) to reach out and request they move to CLIENT, I'm hoping the app might be able to help address this in a future update.

Perhaps when selecting the ROUTER role (each time), and upon connection (one time), if your node is in a ROUTER role, a modal appears:

"Your node is currently in a role that could negatively impact the mesh health of your area.  Please confirm that this node is at the highest possible elevation and has a 360 view of the surrounding terrain.  If not, others will not like you very much."

With the button defaults reverting things to CLIENT unless they check a box and do a dance.  Maybe some links to further reading as well.

 - https://meshtastic.org/blog/choosing-the-right-device-role/
 - https://www.youtube.com/watch?app=desktop&v=htjwtnjQkkE

I suppose this isn't Android-client specific, now that I consider it, but perhaps it's a good starting place.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

shortwavesurfer2009, 2025-02-28:
Yes, please. My thought would be something to the effect of, is your node at least 100 feet above ground level.

shortwavesurfer2009, 2025-02-28:
If you can't reach out and contact them through the mesh, you could do war driving by getting a dummy load for 915 megahertz or 868 megahertz and attach that to your node and then go track down the signal manually and knock on their door.

mahlonsmith, 2025-02-28:
> If you can't reach out and contact them through the mesh, you could do war driving by getting a dummy load for 915 megahertz or 868 megahertz and attach that to your node and then go track down the signal manually and knock on their door.

*knock knock*   "Hello, this is the role police.  Open up!!"   😆 

I suspect as popularity grows, this sort of thing will continue happening at increased frequency -- excited newcomers that aren't doing enough research.   Short of having a local team of spare time wardrivers, I think some "best practice" hinting in the clients would go a long way to help encourage good habits.   👍

shortwavesurfer2009, 2025-03-01:
> _knock knock_ "Hello, this is the role police. Open up!!" 😆
>
> I suspect as popularity grows, this sort of thing will continue happening at increased frequency -- excited newcomers that aren't doing [...]

I definitely agree the configuration spots should have some barriers to making your node a router.

infered5, 2025-03-01:
This is also a problem with my area. A well placed router_late can do some mitigation, but an ounce of prevention is worth a pound of cure.

erayd, 2025-03-04:
An alternative (or complementary) proposal for addressing this is here: https://github.com/meshtastic/firmware/issues/5851

My personal preference is to have it behind remote admin, rather than just a click-through box, because people ignore click-through boxes, and remote admin serves to ensure at least a minimal level of competence and a remote way to fix the problem they caused by picking the wrong role, if that turns out to be necessary.

mahlonsmith, 2025-03-04:
> An alternative (or complementary) proposal for addressing this is here: [meshtastic/firmware#5851](https://github.com/meshtastic/firmware/issues/5851)

Complimentary, I think -- but a fantastic idea.  ROUTER can be disruptive enough that the "are you sure?" is a nice final check, but having the roles only appear via remote admin would be a much surer safety rail.  Love it.

shortwavesurfer2009, 2025-03-04:
I'm good with whatever makes it much more difficult for standard users to bumble into hurting the mesh. locking it behind remote admin seems like a pretty fair trade since the vast majority of users not be enabling remote admin unless they are deploying the device in a location they cannot easily get to. plus remote admin means having to deal with public and private keys and takes more competence to get set up.

mahlonsmith, 2025-03-04:
... it also means a requirement of at least 2 devices, which immediately removes someone first dipping their toes into Meshtastic being an accidental chaos agent.

PR #1636 (2025-03-05): feat #1611: Add confirmation dialog when changing to router role
Introduced a confirmation dialog that will be shown when changing the device role to "router". The confirmation dialog includes links to the "Device Role Documentation" and "Choosing The Right Device Role" blog post. The user must check a checkbox to confirm.

resolves #1611 

[image]
closed 2025-03-05 by jamesarich