meshtastic/Meshtastic-Android #1613: [Missing Translations]: Most config item Strings are hardcoded vs using translateable resource strings

jsoberg, 2025-02-25:
### Contact Details

myheartpills on discord

### Tell us your idea.

I noticed that the code that drives most of the list items in the Radio Configuration screen ("User", "Channels", "Device", etc) are using hardcoded strings (in `ConfigRoute`,  `ModuleRoute`, etc). Moving these configs to instead use String resources would allow us to eventually translate them.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jsoberg, 2025-03-02:
Also noticing that most of our preference names (currently using `EditTextPreference`) are using hardcoded English vs translatable strings.

dkastl, 2025-04-25:
I also realized that these settings were not available in Crowdin. Is there any other reason that they are not localized beside the effort needed?

PR #1794 (2025-04-28): refactor (i18n): extract string resources from config for translation
The strings for config screens are now retrieved from string resources instead of being hardcoded.

resolves #1613
closed 2025-04-28 by jamesarich