meshtastic/Meshtastic-Android #1615: [Bug]: App cannot set TX power (dBm) to negative numbers

NomDeTom, 2025-02-25:
### Contact Details

NomDeTom on Discord & Github

### What happened?

The firmware allows negative values of TX power, and for some amplified nodes this is important to remain compliant or maintain consistent transmission strength. The android app does not allow these values to be entered.

### App Version

2.5.17

### Phone

Samsung Galaxy S20

### Device

Promicro DIY with Seeeeeeed WIO SX1262

### Firmware

2.5.22

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

battlehax, 2025-02-26:
To add to this, it appears the client is handling the tx power value as an unsigned int, and when configuring or remote admin a device which is set negative, the TX power is shown as 4,294,967,295 or a couple less.

NomDeTom, 2025-02-26:
Just ran it through the CLI, and got the following:
```femto@femtofox:~$ meshtastic --set lora.tx_power -2
No Serial Meshtastic device detected, attempting TCP connection on localhost.
Connected to radio
Set lora.tx_power to -2
Writing modified preferences to device
femto@femtofox:~$ meshtastic --get lora.tx_power
No Serial Meshtastic device detected, attempting TCP connection on localhost.
Connected to radio
lora.tx_power: -2
Completed getting preferences

NomDeTom, 2025-02-26:
> To add to this, it appears the client is handling the tx power value as an unsigned int, and when configuring or remote admin a device which is set negative, the TX power is shown as 4,294,967,295 or a couple less.

Yep:

[image]

PR #1629 (2025-03-02): fix #1615: Add explicit signed int editing preference
Fixes #1615

Adds an explicit edit text preference for signed integers, and moves TX power to this preference so that negative values can be set.

| Before | After |
|------|-----|
| <video src="https://github.com/user-attachments/assets/8e80f931-6b13-4a61-8109-86b045cd966c" width="300"></video> | <video src="https://github.com/user-attachments/assets/93da5597-0425-4095-9858-7abc678a645e" width="300"></video> |
commit d8f67e01 (2025-03-02): fix #1615: Add explicit signed int editing preference (#1629)
closed 2025-03-02 by jamesarich