meshtastic/Meshtastic-Android #1616: [Bug]: Map labels seizing issue

canadamadman, 2025-02-26:
### What happened?

the node labels on the map are incorrect size and distance from the node they are attached to. the txt is verry large taking up a lot of visual space and the label is very far away from the node they represent.
 i have pics but don't know how to ad them here.

### App Version

2.5.17

### Phone

tablet version. 

### Device

all

### Firmware

all

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

prokrypt, 2025-02-26:
I will attach screenshots of the app on my tablet as well.

[image]
[image]

madeofstown, 2025-02-26:
Didn't see what Android version that is. Can you try installing the fdroid package instead of the Google play package and see if there is a difference for you?

canadamadman, 2025-02-26:
i can not. devices are under warranty and that need security bypass to install it. sorry.

prokrypt, 2025-02-26:
Some more samples, same tablet, different "Smallest width" settings...

[image]
[image]

canadamadman, 2025-02-26:
so its huge because its small? -huh-

canadamadman, 2025-02-26:
never in a million years would i have thought to do that

canadamadman, 2025-02-26:
i changed what i could to big and no change. what did you change?

prokrypt, 2025-02-26:
> i changed what i could to big and no change. what did you change?

In android developer options, I changed the "Smallest width" setting. My usual number is 1028, but I also tried 1920 (tiny font/icons), and ~400 (huge font/icons). This is an issue with the android meshtastic app font scaling though, so you shouldn't bother trying to change that setting...

madeofstown, 2025-02-27:
> i can not. devices are under warranty and that need security bypass to install it. sorry.

That's definitely not the case. All you need to do is allow installation of APKs from not the play store. It won't void your warranty.

jsoberg, 2025-03-02:
This is also an issue of not appropriately resizing the marker font for your device's font size settings. I think I know what the issue is here, I'll put up a PR

commit 5da82747 (2025-03-02): fix #1616 (a11y): Use SP font size for marker labels on the map (#1630)

* Use and SP font size for marker labels on the map

* Fix detekt issue
commit 749c4b49 (2025-03-04): fix #1616: Calculate DP to pixel value for map label offset (#1637)
closed 2025-03-02 by jamesarich