meshtastic/Meshtastic-Android #1621: [Feature Request]: Reset All button

Ccz119, 2025-02-26:
### Tell us your idea.

Hi all. How about a button or a choice in menu to reset all settings in app so to restart anew.
Thank you all.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-02-27:
This is what the factory reset does

todd2982, 2025-02-28:
If you are looking for app specific settings outside of the radio, there really isn't any.  Your list of nodes you connected to in the past is handled by your phone's Bluetooth pairing in your phone settings.  Outside of that you can lookup how to clear the application's storage (not cache) for your particular phone.

Most Android apps don't have a reset to default option, and instead point you to clear the app's storage.

shortwavesurfer2009, 2025-02-28:
Just clear the app storage. Open the app. Then open the app switcher. Then tap the app icon and tap info. And then it will bring you to your system settings where you can tap storage and then tap clear storage.

jamesarich, 2025-02-28:
closing - can be achieved by clearing app storage in settings

closed 2025-02-28 by jamesarich