meshtastic/Meshtastic-Android #1624: [Feature Request]: Adding new coloumn

jibinbabuamakkattu, 2025-02-28:
### Contact Details

amakkattuj@gmail.com

### Tell us your idea.

I mirrored the same and added a column named temperature. (as mentioned in a previous issue)
fileToWrite.println(
"time,from,sender name,sender lat,sender long,rx lat,rx long,rx elevation,rx snr,distance,hop limit,temperature,payload");
but the problem is that, still when I download the csv, it doesn't contain the new header.
Know why it's happening?

And another doubt I had is that does it create a new header if the old rangetst.csv exists from the previous sessions? like I've been testing it out, and even after I flash a new firmware and download the csv, the data from the old csv exists in it. could it be the reason that the new header isn't being written? if yes, how do I rectify it?

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-02-28:
The range test file on Android is not generated by the firmware, it is based on the debug log data.

jibinbabuamakkattu, 2025-02-28:
> The range test file on Android is not generated by the firmware, it is based on the debug log data.

So i cant add a new coloumn?

jibinbabuamakkattu, 2025-02-28:
> The range test file on Android is not generated by the firmware, it is based on the debug log data.

Sorry if it sounds dumb, but im new to this and i need to add a new coloumn. How can i do it? Any help on the matter would be great

jibinbabuamakkattu, 2025-02-28:
> The range test file on Android is not generated by the firmware, it is based on the debug log data.

When using the network(wifi ssid and password)the firmware creates the csv right?

closed 2025-05-24 by github-actions[bot]