meshtastic/Meshtastic-Android #1632: [Bug]: Accessibility: a few more incorrectly labeled elements when using talkback

shortwavesurfer2009, 2025-03-02:
### What happened?

When using the TalkBack screen reader on Android, the following buttons are not labeled properly.

[image]
[image]

This checkbox says nothing at all.

[image]

And this button reads share when it really should say copy link.

[image]

### App Version

2.5.17

### Phone

OnePlus Nord N200

### Device

Na

### Firmware

Na

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

todd2982, 2025-03-21:
I'd like to work on this one.

shortwavesurfer2009, 2025-03-21:
I see no reason you can't. People submit PRs often and get them accepted

todd2982, 2025-03-21:
- [X] Node sort
- [ ] Map current position
- [ ] Channel selected to share
- [ ] Copy channel URL

shortwavesurfer2009, 2025-03-21:
Keep in mind the location button has two separate states. But otherwise, yes. Those labels sound good.

todd2982, 2025-03-23:
The check boxes for the channels in the QR code screen read back checked/unchecked status in the emulator.  The other items have been corrected.  I'll submit the PR shortly.

jamesarich, 2025-03-24:
fixed by #1705

closed 2025-03-24 by jamesarich