meshtastic/Meshtastic-Android #1634: [Bug]: Pax display on node list solved

Donnie650, 2025-03-03:
### Contact Details

donmbt@gmail.com

### What happened?

I found the error hat stops the pax from displaying on node list.
Node.kt file;
    private fun PaxcountProtos.Paxcount.getDisplayString() = "PAX: ${ble + wifi} (B:$ble/W:$wifi)".takeIf {  ble != 0 || wifi != 0 }
Not
    private fun PaxcountProtos.Paxcount.getDisplayString() = "PAX: ${ble + wifi} (B:$ble/W:$wifi)".takeIf {  ble != 0 && wifi != 0 }

This caused it to not display if only bleutooth or wifi was detected. which is what I kept experiencing.
I will not PM as I not up on doing it. please fix for the others would need it also.

### App Version

2.5.17

### Phone

Android 23

### Device

All

### Firmware

2.6.0

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Donnie650, 2025-03-03:
[image]

jamesarich, 2025-03-03:
good catch, easy fix.

Donnie650, 2025-03-03:
I am glad to have helped you all. Love meshtastic and my Pax works. 💯 :)

jamesarich, 2025-03-03:
> I am glad to have helped you all. Love meshtastic and my Pax works. 💯 :)

Feel free to give the build artifact generated by that PR a try to see if it is working as expected.

PR #1635 (2025-03-05): fix: Update Paxcount display condition
Modify the Paxcount display logic to show the string if either `ble` or `wifi` is not zero, rather than requiring both to be non-zero.

fixes #1634 #1552 

thanks for the callout @Donnie650 !
commit 68cd540f (2025-03-05): fix #1634 #1552: Update Paxcount display condition (#1635)

Modify the Paxcount display logic to show the string if either `ble` or `wifi` is not zero, rather than requiring both to be non-zero.
closed 2025-03-05 by jamesarich