meshtastic/Meshtastic-Android #1638: [Feature Request]: Do not show battery as full charge, just because the node is connected to power

shortwavesurfer2009, 2025-03-04:
### Tell us your idea.

As demonstrated in the screenshot below, as soon as you connect a node to power, it immediately jumps from whatever the battery charge was to 100% even though the battery is not fully charged.

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

prokrypt, 2025-03-04:
I believe it may be the node firmware that sends the 100% (or 101%?) reading, but I agree. Maybe the voltage can be graphed displayed along with the percentage so we can tell when charging starts/stops.

shortwavesurfer2009, 2025-03-28:
An update to this. Looks like they cant fix it in firmware. https://github.com/meshtastic/firmware/issues/6399

garthvh, 2025-03-28:
It is not 100 it is 101 which indicates it is powered, this is working as intended

closed 2025-03-28 by garthvh