meshtastic/Meshtastic-Android #1640: [Feature Request]: add waypoint marker without sending

prokrypt, 2025-03-04:
### Tell us your idea.

Would be nice if we also had a "save" or "mark" button for the waypoint dialog.

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

todd2982, 2025-03-27:
It makes sense to me to get this implemented in firmware first so it can be the source of record for the locally stored waypoints.  This would also allow having the same data when switching between client applications.  I expect NRF vs ESP nodes to have different limits on storage.

Thoughts?

garthvh, 2025-03-27:
Waypoints are a transient packet that is not cached in the firmware.

todd2982, 2025-03-27:
Good to know.