meshtastic/Meshtastic-Android #1654: [Bug]: F-Droid build failed

linsui, 2025-03-07:
### What happened?

In https://github.com/meshtastic/Meshtastic-Android/pull/1607 the awesome-app-rating lib which was only a dep of the google flavor is now a dep for all flavors. This seems to be a mistake. Could you please fix it? Thanks!

### App Version

2.5.18

### Phone

Android

### Device

Android

### Firmware

Android

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-03-07:
good catch, thank you

IzzySoft, 2025-03-07:
Thanks @linsui for beating me here – and @jamesarich for the prompt fix! Is a new release planned soon? Asking to choose the correct path at IzzyOnDroid 😉

jamesarich, 2025-03-07:
We can definitely break off another release if it's blocking @IzzySoft - let me know!

IzzySoft, 2025-03-07:
It's blocking F-Droid. At IoD I've just (temporarily) added the `NonFreeComp` (proprietary components) and `NonFreeNet` anti-features. Our scanner will inform me when the two libs (Play Core + GMS) are gone, so I'll remove them again then. If you need a little more time, I'd also add the translation for the warning – but I guess it won't be longer than a day or two, so it's not worth it… 🙈

jamesarich, 2025-03-07:
We'll get our few PRs wrapped and another build cut hopefully in a day or so. 🙊

IzzySoft, 2025-03-07:
Wonderful, thanks!

jamesarich, 2025-03-08:
@IzzySoft https://github.com/meshtastic/Meshtastic-Android/releases/tag/2.5.19 should hopefully solve your woes

IzzySoft, 2025-03-08:
Builder is still running, but peeking into the (not yet committed) logs, I can confirm:

```json
        "reproducible": true,
        "error": null,
```

Thanks a lot!

PR #1656 (2025-03-07): fix #1654: Move the app rating dependency to googleImplementation
The `awesome.app.rating` dependency is now under `googleImplementation` instead of `implementation` in `app/build.gradle`.
fixes #1654
PR #1659 (2025-03-08): 2.5.19
This increments the app's internal version code and the user-facing version name in the build.gradle file.

Should fix #1654
commit d40672df (2025-03-11): fix #1654: Move the app rating dependency to googleImplementation (#1656)

* Move the app rating dependency to googleImplementation

The `awesome.app.rating` dependency is now under `googleImplementation` instead of `implementation` in `app/build.gradle`.

* Refactor alert DND dialog padding

Use `dpToPx` for padding in the alert DND dialog for better consistency.