meshtastic/Meshtastic-Android #1662: [Feature Request]: Offline Node Notifications

tggege, 2025-03-09:
### Tell us your idea.

Hello, I would like the Android application to alert us when a node no longer gives news after a certain defined period. On the same principle as the battery level.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Kealper, 2025-03-18:
This is something that's already on my to-do list, "online" and "offline" notifications for nodes marked as favorites.

Eventually™

todd2982, 2025-03-27:
This sounds like your phone could get flooded with messages if you travel outside of your local mesh.  Please make sure this is opt-in when working on it.

Kealper, 2025-03-27:
> This sounds like your phone could get flooded with messages if you travel outside of your local mesh. Please make sure this is opt-in when working on it.

Yeah when I do end up adding it, it would be as a separate notification type that would be opt-in because of the amount of notification spam it could (would) cause. Down the road, setting specific nodes to have this enabled or disabled would be ideal, instead of just a blanket "is node favorite?".

closed 2025-05-24 by github-actions[bot]