meshtastic/Meshtastic-Android #1663: [Bug]: Partial reconnect over bluetooth once device has rebooted

arthurlutz, 2025-03-09:
### What happened?

Hi, started using Meshtastic android app with a heltech v3 device, first time I pair using bluetooth (with the code) I can see all the settings and change them, but as soon as the device reboots I can't reconnect to it, I have to go in the Android Settings and forget the device and redo the code... Is this a feature ? A known bug ? Am I doing things wrong ? 

### App Version

 2.5.17

### Phone

/e/os 2.0 Android 13

### Device

heltech v3

### Firmware

2.5.20

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

todd2982, 2025-03-24:
Did you pair to the radio with the Meshtastic app, or through the phone OS?

arthurlutz, 2025-03-24:
Through the Meshtastic app