meshtastic/Meshtastic-Android #1665: [Bug]: traceroutes that don't happen still get logged

prokrypt, 2025-03-10:
### What happened?

A bug happened!

When a traceroute gets denied due to the 30 second limit, it still gets logged.

[image]

### App Version

2.5.17

### Phone

a potato

### Device

nano g2 ultra

### Firmware

2.5.21

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Dtrieb, 2025-03-12:
I think, it would be an idea to get a toast message that there is no response to this Node.

jamesarich, 2025-03-13:
> I think, it would be an idea to get a toast message that there is no response to this Node.

@Dtrieb This seems unrelated to this issue but may have merit as it's own feature-request [issue](https://github.com/meshtastic/Meshtastic-Android/issues). Feel free to drop one in.

jamesarich, 2025-03-13:
>When a traceroute gets denied due to the 30 second limit, it still gets logged.

@prokrypt I think it still makes sense that the attempt is logged.  The messaging could be changed to `denied` or something to that effect, so that it is differentiated from the actual `no response` logs.

closed 2025-05-24 by github-actions[bot]