meshtastic/Meshtastic-Android #1670: [Bug]: After a restart of the smartphone Meshtastic App didn't survive

Dtrieb, 2025-03-12:
### Contact Details

Dtrieb

### What happened?

A bug happened!

If I restart my smartphone the smartphone Meshtastic App didn't survive. I have manual start it again, and connect to the Node manually.

### App Version

2.5.19

### Phone

Pixel 7 Pro.

### Device

Tracker 1000-e

### Firmware

2.6.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-03-13:
@Dtrieb can you reproduce the issue consistently?

Dtrieb, 2025-03-13:
> [@Dtrieb](https://github.com/Dtrieb) can you reproduce the issue consistently?

Yes every reboot it happens.

todd2982, 2025-03-24:
I found this as a possible reason, and way forward.  If somebody else agrees, I can work on a PR.
https://stackoverflow.com/questions/72972225/android-app-to-start-automatically-on-boot-up

Dtrieb, 2025-06-19:
> I found this as a possible reason, and way forward. If somebody else agrees, I can work on a PR. https://stackoverflow.com/questions/72972225/android-app-to-start-automatically-on-boot-up

Nobody agrees, but nobody do something, so can You give it a try?