meshtastic/Meshtastic-Android #1675: If Traceroute didn't respond to a note get a toast message.

Dtrieb, 2025-03-13:
> I think, it would be an idea to get a toast message that there is no response to this Node. 

 _Originally posted by @Dtrieb in [#1665](https://github.com/meshtastic/Meshtastic-Android/issues/1665#issuecomment-2717350519)_

prokrypt, 2025-03-13:
We already get a popup when there's a response. However, when do we decide that no response was received? 30 seconds? 5 minutes?

In the past, I've had traceroutes that returned after about an hour...

prokrypt, 2025-03-13:
Possible similar idea: some sort of notification when a traceroute fails to go out at all (sort of like the "max retries reached" for messages)

shortwavesurfer2009, 2025-03-20:
What causes Traceroute to take so long anyway? An hour seems absolutely ridiculous. Personally, I figure if it doesn't respond in 30 seconds, then I consider it to have failed. But that's just me.

jsoberg, 2025-04-18:
Linking to https://github.com/meshtastic/Meshtastic-Android/issues/1765

closed 2025-05-27 by github-actions[bot]