meshtastic/Meshtastic-Android #1680: [Feature Request]: PMSA003 Display sensor data in the app

MALAONE1, 2025-03-14:
### Tell us your idea.

Is it possible to display the PMSA003i sensor data under Environment (in App Details)in the future? This sensor has been supported by Meshtastic for some time. From RAK, it's the RAK12039.

[image]

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jamesarich, 2025-03-14:
it looks like we'd want to expose [AirQualityMetrics](https://buf.build/meshtastic/protobufs/docs/main:meshtastic#meshtastic.AirQualityMetrics) as we do the other telemetry metrics types.

garthvh, 2025-03-14:
> it looks like we'd want to expose [AirQualityMetrics](https://buf.build/meshtastic/protobufs/docs/main:meshtastic#meshtastic.AirQualityMetrics) as we do the other telemetry metrics types. It is a different metric than `Environment` metrics so probably should have it's own log/graphs.

There are some more details here, but in general point in time readings of the PMS sensor is not really human readable without some data stored in the app and some calculations run in order to generate AQI useful (Air Quality Index) numbers. (different that the IAQ indoor air quality number provided by a BME680) You need about 50 samples over 3hours to get a meaningful number for end users.

MALAONE1, 2025-03-14:
@garthvh The data in PM1.0, PM2.5, and PM10.0, and fine particulate matter per 0.1 l of air, categorized into 0.3um, 0.5um, 1.0um, 2.5um, 5.0um, and 10um, which this sensor can detect, would be fine and wouldn't necessarily need to be converted to IAQ. The BME is sufficient for that. Especially in areas with a risk of forest fires or similar, the data directly in the app would be very useful.