meshtastic/Meshtastic-Android #1716: [Bug]: Node map is zoomed in way to far by default

shortwavesurfer2009, 2025-03-24:
### What happened?

When you open a node details screen and click on node map, the map that comes up is far too zoomed in by default.

https://github.com/user-attachments/assets/6e0918fb-56b7-42ef-a78e-ef9cf022d1b3

### App Version

2.5.20

### Phone

OnePlus Nord N200

### Device

Na

### Firmware

Na

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

shortwavesurfer2009, 2025-03-24:
So I think you're only going to get another position packet if it moves. For example, there's a node in a fixed location, and it said it had given me its position at 3am this morning, and I just requested it again. And instead of giving me a second position packet, it just updated the current one to the current time. Therefore, the node map does not scale properly. So it seems like unless the position moves, you won't get another position packet in order to properly scale the map.

prokrypt, 2025-03-24:
Maybe set a limit on the max initial zoom so it won't be so disorienting...

b8b8, 2025-03-25:
> So I think you're only going to get another position packet if it moves. For example, there's a node in a fixed location, and it said it had given me its position at 3am this morning, and I just [...]

Yea so maybe for all fixed nodes (or other one position logs) force zoom level to say 1km.

b8b8, 2025-03-25:
> Maybe set a limit on the max initial zoom so it won't be so disorienting...
> 
> The initial zoom with multiple positions could use some adjustment too, as it's way too zoomed out...
> 
> [image]

the multiple positions zoom level seems to leave ~1/3 of screen clearance around the entire cluster regardless of actual distance between position points.

closed 2025-05-23 by jamesarich