meshtastic/Meshtastic-Android #1725: [Feature Request]: Offline map caching

lewi3069, 2025-03-26:
### Contact Details

lewi3069@gmail.com

### Tell us your idea.

I'd like the app to allow the pre-download of an map area. Many times I'm using the app I don't have service and it would be helpful to coordinate with others and their location. At times I can't load the map to view their location relative to mine.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-03-26:
This already exists, although the tiles do eventually time out after a month or two and must be redownload.

[image]

todd2982, 2025-03-27:
> ### Contact Details
> [lewi3069@gmail.com](mailto:lewi3069@gmail.com)
> 
> ### Tell us your idea.
> I'd like the app to allow the pre-download of an map area. Many times I'm using the app I don't have service and it would be helpful to coordinate with others and their location. At times I can't load the map to view their location relative to mine.
> 
> ### Relevant log output
> ### Code of Conduct
> * [x]  I agree to follow this project's Code of Conduct

Options exists as mentioned in previous comment.  There are limits to the map layers that are available to download for offline use.  The default "OpenStreetMap" is not available for download, but if you switch to "USGS TOPO" (and others) you'll see a download button in the bottom right.

Can we close this ticket for you?

lewi3069, 2025-03-27:
Yes, please close. My apologies I hadn't looked for the feature in the other layers.